arduino/docs-content #123: add that zero ohm resistors for USB HS are missing on Portenta Breakout board

Flydroid, 2022-03-28:
Hi,

While using the portenta breakout board we have noticed that there are two zero ohm resistors missing to connect the USB HS from the portenta to the pinout on the breakout board. 
Its not clear why those are missing - maybe just forgotten in the production documents?

on the board - no resistors mounted!
[image]

on schematic
[image]

marqdevx, 2022-03-29:
Hey! @Flydroid Thanks for letting us know!

Your board is fine, Im going to update the schematics and documentation, the schematics should show the following:
[image]

I will create a PR and make you reviewer, feel free to give us your feedback! :D

In case you want to connect those pads, you can do it by soldering a 0ohm resistor or a wire/copper.

Flydroid, 2022-03-29:
Awesome!
But is there a reason why it's not mounted? Interference when using the normal USB connector? 

Yes that's we did!

marqdevx, 2022-04-07:
PR ready #153

pnndra, 2022-04-08:
@Flydroid sorry for the slow reply. the reason why those jumper resistors are not mounted is that those USB lines are the same that go to the on board USB-C connector. if we had those mounted they would create a stub on the traces and may interfere with normal USB-C operation. 
if you intend to use that port from the breakout you have to short those pads AND use the USB only from breakout, leaving USB-C unconnected.

marqdevx, 2022-04-08:
Merged, closing the issue.
Thanks for the report!

closed 2022-04-08 by marqdevx