arduino/docs-content #131: Nano RP2040 documentation has outdated PCB layout

jdfin, 2022-03-31:
I think the root of the problem is the PCB layout is different in these two places:
1. https://docs.arduino.cc/hardware/nano-rp2040-connect, the "Interactive Viewer" pulls up the Altium viewer with (what I think is) outdated files
2. Same page, downloading "CAD Files" then loading those in Altium viewer shows a different PCB layout (which matches my fairly-new board).

An easy-to-spot difference is the IMU (what I was looking for). Clicking "interactive viewer" shows it near the reset button (by the D8 and D9 pins). That's not the IMU on my board (not sure what it is). Downloading the files and loading them into Altium viewer shows it near the A5/SCL pin, which matches what I see on my board.

If the IMU (on new boards) is as shown in the downloadable CAD files, then the "Interactive Viewer" should be fixed to load those files. Also, in the "cheat sheet" page (https://docs.arduino.cc/tutorials/nano-rp2040-connect/rp2040-01-technical-reference), the picture identifying the IMU should be fixed (it is highlighting something else based on (what I think is) the wrong PCB file. On the other hand, if I happened to get an old board and the IMU on new boards is up by D8 & D9, then the "Download CAD Files" link should be fixed.

The reason this could matter, at least a tiny bit, is that I want to put the IMU at the center of rotation of my device to simplify calculations. It only moves about 10mm between the two, but I dunno, I may as well try. :)

karlsoderby, 2022-04-04:
Hi @jdfin! We are now working to update these files, thank you reporting it!

marqdevx, 2022-04-19:
Fiiles updated on #177 

@karlsoderby We need to request the new pictures. Feel free to merge my PR.

marqdevx, 2022-04-19:
the IMU screenshots should be updated the correct IC: [image]

karlsoderby, 2022-04-21:
@jdfin the files are now up to date, so I am closing this issue. Please, if some issue persist, feel free to re-open it. Have a great day.

closed 2022-04-21 by karlsoderby

marqdevx, 2022-04-21:
Note: I didnt request the IMU picture

jdfin, 2022-04-21:
Nice! The 3d viewer for the pcb is a neat feature when you need it.

I noticed that if I click the "CAD Files" button under "Downloads" now, I get a zip file that seems to only have a .step file in it. I can open that in Fusion 360 and it brings up a model of the board (and maybe it works elsewhere), so maybe that's correct, or correct enough; just thought I'd point out it's different from before when I got a zip with .pcbdoc and .schdoc files in it which would drop into altium viewer.

What a neat board. Add a kitchen sink and it'll be complete.