arduino/docs-content #134: Possible wrong RGN sequence

tahunus, 2022-04-03:
In Line 128 of Arduino_MKRIoTCarrier.h the LED instance is created with: 
` Adafruit_DotStar leds = Adafruit_DotStar(NUMPIXELS, DATAPIN, CLOCKPIN, DOTSTAR_BGR); `

link to file : [https://github.com/arduino-libraries/Arduino_MKRIoTCarrier/blob/master/src/Arduino_MKRIoTCarrier.h](url)

Which means the color sequence is BGR, not RGB, which means that the MKR IoT Carrier Cheat Sheet has an error in 
[https://docs.arduino.cc/tutorials/mkr-iot-carrier/mkr-iot-carrier-01-technical-reference#leds](url)

Where the picture lists how to get the colors using the RGB order
[https://docs.arduino.cc/static/e66517362e06a9efb932279bbf51e6dc/29114/mkrIotCarrier-leds.png](url)

Then the code below it suggest the sequence green, red, blue:
`carrier.leds.setPixelColor(index, green, red, blue);`

Which works correctly, although it's not the BGR of line 128 of MKRIoTCarrier.cpp

Total confusion

marqdevx, 2022-04-06:
We will make a PR on the MKRIoTCarrier library to make the LEDs be RGB as it is more common.

marqdevx, 2022-05-25:
Update: Library is OK
I will make changes on our documentation to show the correct order, which is RGB

marqdevx, 2022-08-25:
Thanks for spotting the issue @tahunus 🙌 
It should be fixed now 😄

closed 2022-08-25 by marqdevx