arduino/docs-content #147: Pro | Updating Bootloader tutorial mentions file that doesn't exist

bashynx, 2022-04-06:
The tutorial for updating bootloader for Portenta board has wrong file mentioned https://docs.arduino.cc/tutorials/portenta-h7/updating-the-bootloader

It seems that in newest Core version, the updatingBootloader file got renamed, at least I presume it's the same file.

[image]

per1234, 2022-04-06:
It was renamed here:

- https://github.com/arduino/ArduinoCore-mbed/commit/7fee0c15cfc2ff9accc4f79d2e121fafbafb7302
  - library name changed from "**Portenta_System**" to "**STM32H747_System**"
  - Example sketch name changed from "**PortentaH7_updateBootloader**" to "**H747_updateBootloader**"
- https://github.com/arduino/ArduinoCore-mbed/commit/8ab4ec43ec88644b1846b4b5130930adbc02d62e
  - Example sketch name changed from "**H747_updateBootloader**" to "**STM32H747_updateBootloader**"

There are additional occurrences of the outdated library and example name, so a comprehensive review should be done of the documentation content rather than patching only this specific tutorial.

marqdevx, 2022-04-06:
Noted and we will update all the content related to that change

Thanks @bashynx @per1234 😄

marqdevx, 2022-04-19:
Fix merged, thanks for the reporting it! :D 
#176