arduino/docs-content #161: getting started exemple isn't compiling

gargoyle777, 2022-04-12:
"
#include "camera.h"

CameraClass cam;
uint8_t fb[320*240];

void setup() {
  Serial.begin(921600);  

  // Init the cam QVGA, 30FPS
  cam.begin(CAMERA_R320x240, 30);
}

void loop() {
  // put your main code here, to run repeatedly:
  
  // Wait until the receiver acknowledges
  // that they are ready to receive new data
  while(Serial.read() != 1){};
  
  // Grab frame and write to serial
  if (cam.grab(fb) == 0) {
     Serial.write(fb, 320*240);       
  }
  
}
"
is the code

"
 error: 'CameraClass' does not name a type; did you mean 'CameraWire'?
 CameraClass cam;
" is the error

gargoyle777, 2022-04-12:
thanks for the answer, i tried that and it's compiling. But i have another problem now: the Java Processing sketch displays a weird distorted image, like if the video was made from orizzontal stripes, not aligned, then it times out. taking out the not ( !) in this line (if(!timeoutDetected && Serial.read() != 1) return;) of the arduino code solves the timeout problem, but not the distorsion one

marqdevx, 2022-04-13:
Hey @gargoyle777 if this issue doesn't come from our docs site, I would suggest to move this "ticket" to the Core repository.
Thanks

per1234, 2022-04-13:
@marqdevx the problem is not in the Core repository.

The problem is with this tutorial:
https://docs.arduino.cc/tutorials/portenta-vision-shield/vs-ard-gs

The source of that is stored in this repository, so the fix must be done in this repository:
https://github.com/arduino/docs-content/blob/241d6b14034eadfc179e00069f538f999a2abc37/content/hardware/04.pro/shields/portenta-vision-shield/tutorials/vs-ard-gs/content.md

marqdevx, 2022-04-13:
I missunderstood it, sorry!
We will make those changes, thanks. :)

per1234, 2022-04-13:
I should clarify that my comment was specific to the `error: 'CameraClass' does not name a type; did you mean 'CameraWire'?` error in the original report. That is caused by the sketch in the tutorial being out of date (the updated sketch is [here](https://github.com/arduino/ArduinoCore-mbed/tree/3.0.1/libraries/Camera/examples/CameraCaptureRawBytes)).

I don't know anything about the problem with the image distortion.

marqdevx, 2022-05-25:
Solved at #252 
Thanks for the report! 😄