arduino/docs-content #170: Incorrect command given for running install.sh in ubuntu

imraghavagr, 2022-04-15:
The below command does not execute "install.sh" file for installing arduino-1.x through ubuntu terminal:
```
assets/install.sh
```
Quick Start Github docs [link](https://github.com/arduino/docs-content/blob/main/content/software/ide-v1/tutorials/Linux/Linux.md).
Quick Start Arduino website [link](http://docs.arduino.cc/software/ide-v1/tutorials/Linux).

The solution to the above problem is:
```
sudo sh assets/install.sh
```
NOTE: 
- Remove 'sudo' keyword if already running terminal with administrative rights. 
- The folder structure of the latest 1.8.19 version does not have 'assets' folder in it and the 'intall.sh' is directly placed in the "arduino-1.8.19" folder. In this case run the following command:
```
sudo sh install.sh
```
<b>I would be very happy to send a pull request related to the changes I mentioned above.</b>

marqdevx, 2022-05-02:
Hey, thanks for the report!
I would replicate the issue this week

And yes feel free to make a PR with those changes! 😄

marqdevx, 2022-05-10:
#196 merged fix, thanks!

closed 2022-05-10 by marqdevx