arduino/docs-content #173: Wrong image for schematic KeyboardAndMouseControl (Short Circuit!)

valuta1995, 2022-04-17:
The schematic diagram on this page shows a direct 5v to gnd short. I suggest it is immediately removed and quickly replaced. The drawing seems to be fine. Only the schematic (amber colored image) seems affected.

https://github.com/arduino/docs-content/blob/main/content/built-in-examples/09.usb/KeyboardAndMouseControl/KeyboardAndMouseControl.md

marqdevx, 2022-04-19:
Thanks @valuta1995 for spotting it!

@jhansson-ard Could you help to fix this?

[image]

jhansson-ard, 2022-04-19:
@marqdevx of course! Will make the PR now! ⭐

valuta1995, 2022-04-19:
It looks great. Thanks for the quick fix!