arduino/docs-content #275: Battery connector spec

John-Page, 2022-06-05:
It is not enough to say the battery connector is a JST type.  There are hundreds of different JST connectors.  All references to this should state that the connector is JST-PH which is the specific type.   If you purchase batteries for this they often come with the wrong kind of JST plug.

marqdevx, 2022-06-08:
Hey @John-Page !
Could you point the board's documentation you are talking about?
Thanks :)

John-Page, 2022-06-08:
Here https://docs.arduino.cc/hardware/mkr-wifi-1010
under tech specs, power, battery connections.

and here
https://docs.arduino.cc/static/fc77c3c3c77d69764ba7773df64c99db/ABX00023-datasheet.pdf
under Chapter 4 (pdf page 8), it fails to mention the connector at all.

In both cases it should be specified at JST PH.  JST is the maker like
Molex.  You have to know it is type PH.     This problem is probably on all
the Arduino boards that have this connector also.

Cheers!

John P

marqdevx, 2022-06-09:
Got it, thanks for spotting it! 😄 
We will take care of saying the correct ID of the connector

karlsoderby, 2022-07-26:
@John-Page the MKR board product pages & tutorials have been updated with the "JST PH" specification. This can be viewed in (#416). Datasheet is not changed yet. Thank you for reporting.