arduino/docs-content #288: Style Guide link in CONTRIBUTING.md broken

gvenzl, 2022-06-11:
The link to the Style Guide in [CONTRIBUTING.md](https://github.com/arduino/docs-content/blob/main/CONTRIBUTING.md#style) refers to https://docs.arduino.cc/hacking/software/ArduinoStyleGuide which no longer exists:

[image]

per1234, 2022-06-27:
@jhansson-ard can this be considered resolved by https://github.com/arduino/docs-content/pull/290, or are you planning to also set up a redirect from https://docs.arduino.cc/hacking/software/ArduinoStyleGuide to https://docs.arduino.cc/learn/contributions/arduino-writing-style-guide?

jhansson-ard, 2022-06-27:
@per1234 there is a redirect in a PR, awaiting review from @marqdevx ⭐

jhansson-ard, 2022-07-14:
This link is now redirecting properly. Thank you for your eagle eyes @gvenzl 👀

closed 2022-07-14 by jhansson-ard

gvenzl, 2022-07-14:
You are welcome! :)