arduino/docs-content #306: Incorrect schematic / diagram

Coding-Badly, 2022-06-26:
The schematic [here](https://github.com/arduino/docs-content/blob/main/content/built-in-examples/02.digital/tonePitchFollower/tonePitchFollower.md) ([as published](https://docs.arduino.cc/built-in-examples/digital/tonePitchFollower)) is not correct.  The description and source code all indicate the speaker is connected to pin 9.

The breadboard diagram is incorrect.  The blue wire from pin 9 does not reach the resistor.

marqdevx, 2022-06-27:
@jhansson-ard Could you please take care of this one?
Just move the pin8 to pin9 😄 
(You can also swap the wires of the speaker for better alignment)
[image]

jhansson-ard, 2022-06-27:
Added a new PR, and tried to make @Coding-Badly a co-author (not sure it worked?)! 

Thanks for the eagles eyes @Coding-Badly 🦅

marqdevx, 2022-06-27:
Yes you did, check it out here https://github.com/arduino/docs-content/pull/310/commits/7ab7c0deec28b542e8f6f744aa99225342229dd4

Coding-Badly, 2022-06-27:
> tried to make @Coding-Badly a co-author (not sure it worked?)!

I have review permission.

> Thanks for the eagles eyes @Coding-Badly eagle

The corrections were spotted by and reported to me by [riopstop21](https://forum.arduino.cc/u/ripstop21).  I am merely the messenger.

closed 2022-06-28 by jhansson-ard