arduino/docs-content #312: [Portenta H7 Full] wrong part number secure element

pnndra, 2022-06-28:
line 

https://github.com/arduino/docs-content/blob/6f57afe6ca4119417af5bf369b21023c61e0b136/content/hardware/04.pro/boards/portenta-h7/datasheets/datasheet.md?plain=1#L50 

shall mention SE050C2, not SE0502

also, at line 
https://github.com/arduino/docs-content/blob/6f57afe6ca4119417af5bf369b21023c61e0b136/content/hardware/04.pro/boards/portenta-h7/datasheets/datasheet.md?plain=1#L400
the description of this part is very short and not that accurate since at the moment we're not providing a library to use it. maybe it's worth copying the feature list from datasheet as we did for ATECC608B

marqdevx, 2022-06-28:
Thanks! We've got that reported while ago too.
Is there any task related to this on the backlog @lunnea @aliphys ?

jhansson-ard, 2022-06-29:
@pnndra we have now made a task with the code TC-231, thanks for reporting this! ⭐

closed 2022-08-17 by aliphys