arduino/docs-content #337: Cannot find PortentaBLE example

gmacario, 2022-07-07:
Hello,

I was unable to find

Examples > Arduino_Pro_Examples > BLE Connectivity on Portenta H7 > PortentaBLE

as documented in <https://docs.arduino.cc/tutorials/portenta-h7/ble-connectivity>

Sources: <https://github.com/arduino/docs-content/blob/main/content/hardware/04.pro/boards/portenta-h7/tutorials/ble-connectivity/content.md>

I am running Arduino IDE 2.0.0-rc8 on Windows with

- Board: "Arduino Mbed OS Portenta Boards" Version 3.1.1 (latest as of 2022-07-07)

Libraries:

- ArduinoBLE by Arduino Version 1.3.1

Did I miss anything?

Thanks in advance

gmacario, 2022-07-07:
After installing library "Arduino_Pro_Tutorials" Version 1.0.4 I could find sketch, although under a different menu

Examples > Arduino_Pro_**Tutorials** > BLE Connectivity on Portenta H7 > PortentaBLE

Maybe it would worth to update the tutorial, mentioning the prerequisite library and adjusting the path of the sketch.

marqdevx, 2022-07-15:
Thanks for spotting it, I've fixed it on the shown PR

closed 2022-07-15 by marqdevx