arduino/docs-content #356: BUS bugs inside the Arduino Nano Every Fritzing part

BastianFie, 2022-07-17:


vanepp, 2022-07-17:
There is a corrected version of this part as well as a new part for the SMD version available here:

https://forum.fritzing.org/t/bugs-in-arduino-nano-every-fritzing-part-v6/16589/2

feel free to use them if you like.

marqdevx, 2022-08-08:
@kengdahl Could you please have a look? 😄

kengdahl, 2023-01-05:
@marqdevx I'm afraid that I don't have the knowledge on how to fix this.

vanepp, 2023-01-23:
I posted a fixed version in the Fritzing forum entry 2 posts above (along with a SMD version of the part.) You are welcome to use them both if you like. You should be able to just copy the .fzpz files from the Fritzing post in to your web site after checking they do what you want.

marqdevx, 2023-01-23:
@vanepp I'll check locally if everything is alright, and commit the changes giving you the authorship.

Thanks for the contribution 😄 
I'll keep you updated.