arduino/docs-content #389: missing documentation and example code: how to quickly and permanently change SPI clock divider

shiftleftplusone, 2022-07-17:
missing documentation and example code: 
how to quickly and simply change SPI clock divider ?
About setClockDivider():
"This function should not be used in new projects. Use SPISettings with SPI.beginTransaction() to configure SPI parameters."

But I don't understand how to use SPI.beginTransaction() or either function  for AVR, and ARM, and ARM extended mode. More source code examples would be needed for this.

jcarolinares, 2024-06-06:
Hello, this is outdated and open for more than 9 years.

I think is a discussion more related at this moment to the arduino forum

@jacobhylen idem, I suggest to close this issue

jcarolinares, 2025-03-13:
@jacobhylen never answered my request.

I'm going to close this issue, thank you all.

closed 2025-03-13 by jcarolinares