arduino/docs-content #450: Learn:The Arduino Guide to Low Power Design - Need to command the 1310 Murata module to sleep?

OakleyNode, 2022-08-06:
Hello all,
I am studying the Low Power Design guide, and I see a paragraph that says:  "It is important to know that the Low Power task applies only to microcontroller. This means that external modules such as Murata module we used here, found on MKR WAN 1310 board, must be coded in the task separately to make the module go into sleep state."

However, I also see in the MKRWAN library, in issues 13 and 28, statements indicating that commanding the Murata module to sleep is not necessary because "...the module will sleep by itself when idle, and there is no need to call sleep() for low power operation" (per Mr. Facchinm = facchinm).

I could easily be missing some key fact, but I wanted to check to see if the documentation should be updated, or if I'm misunderstanding the point being made in those two issues in MRKWAN library.

thank you in advance.

marqdevx, 2022-08-08:
Hello!
I guess you are talking about this part of the tutorial https://docs.arduino.cc/learn/electronics/low-power#lora-transmitter-with-low-power-example

The sketch is showing how to control low power commands, to tell the board and its module to sleep
```
// Low Power Library
#include "ArduinoLowPower.h"

// LoRa Library
#include <SPI.h>
#include <LoRa.h>

// LoRa Packet Content 
char* message = "Hello LoRa!";

void setup() {
  Serial.begin(9600);
  while (!Serial);

  // LoRa Setup
  Serial.println(F("LoRa Sender"));
  if (!LoRa.begin(868E6)) {
    Serial.println(F("Starting LoRa failed!"));
    while (1);
  } else {
    Serial.println(F("Starting LoRa Successful!"));
  }
}

void loop() {
  LoRa_Packet_Sender();
  GoToSleep();
}

// LoRa Task
void LoRa_Packet_Sender() {
  Serial.print(F("Sending packet: "));
  Serial.println(message);

  // send packet
  LoRa.beginPacket();
  LoRa.print(message);
  LoRa.endPacket();

  // Putting LoRa Module to Sleep 
  Serial.println(F("LoRa Going in Sleep"));
  LoRa.sleep();
}

// Sleep Task 
void GoToSleep(){
  Serial.println(F("MKR WAN 1310 - Going in Sleep"));
  LowPower.deepSleep(20000);
}
```

The LoRa has this command https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L129

Issues:
* https://github.com/arduino-libraries/MKRWAN/issues/14
* https://github.com/arduino-libraries/MKRWAN/issues/28

It seems Murata's firmware has that function handled automatically https://github.com/arduino/mkrwan1300-fw

@canchebagur @TaddyHC Could you please check that out? and if so add a note about that maybe? 🤔 

The only concern I have its that those issues are from 2018, and I don't know the latest firmware versions on Murata's module @OakleyNode but lets check it out!

OakleyNode, 2022-08-08:
Hi @marqdevx -- thank you for your response!

I was more looking at that narrative section (that I cited) in the guide, and not so much at the example you're showing, but I think it raises the same question, i.e. do I need to include LoRa.sleep() in my code, or is it already being done automatically by the module, and so LoRa.sleep() is not needed?

About your question: I don't know the latest firmware versions on Murata's module @OakleyNode:
--> I'm using:  **ARD-078 1.2.3** on a MKRWAN 1310 (s/n 07291).

marqdevx, 2022-08-08:
Yeah just made a summary to know the context 😄 
Lets see if my colleagues can help us out!

TaddyHC, 2022-08-08:
Hello all!

In this case, as you quoted, it is not explicitly required to include LoRa.sleep() inside the code. As the Murata module will get into the Idle state handled by the firmware. But for this example, we have included explicitly the LoRa.sleep() to put the Murata module to sleep state, controlling the module's state by ourselves. This gives us more control and would be independent of the tasks handled by Murata module, as even if it finishes the packet transmission, the module may stay a certain amount of time without going into a low power state. 

So the idea is to make sure the device goes into sleep state when we task them to, without depending on the module's present load condition. With this, we control the device's behavior and gain a bit of the lifetime of the device for power-critical applications. It is an extra step and usually, it is possible to let the modules control the power state itself, but for the best result on power-critical applications, it is a nice detail to take into account.

Thus, LoRa.sleep() is not a must-have method and the Murata module will automatically put itself to idle state given load conditions. **However**, as we are discussing low power design, the idea is to have the control over device's power state and reduce power consumption as much as we can. Therefore, LoRa.sleep() is used to put the Murata module to sleep state explicitly and at designed instances as shown in the example shared above. 😄

marqdevx, 2022-08-15:
@OakleyNode What do you think? IMO there is no change required, let me know! 👀

OakleyNode, 2022-08-15:
Hi all,

Sorry for my delayed response, I missed @TaddyHC's reply on the 8th.  It sounds like using LoRa.sleep, while not absolutely necessary, is a good practice, or a best practice.  And it ensures that "positive control" is being exercised by the code. And assuming there are no other drawbacks, then I agree no change is required to the documentation.

I'm very new to Arduino and coding and try to not to take anything for granted, so I appreciate you all taking the time to review and respond. I will mark this issue as closed / resolved. Thanks again!

cheers,
               ...Norm...

closed 2022-08-15 by OakleyNode