arduino/docs-content #495: Information about Braccio++ smart servo motors

metanav, 2022-08-22:
I am trying to drive the Braccio++ using ROS2. I need following information for the URDF.

- Maximum joint speed (velocity in rad/s)
- Maximum joint acceleration (velocity in rad/s^2)
- Maximum joint torque (effort in N-m)

Where can I find the datasheet for the smart servos?

marqdevx, 2022-08-30:
Did you check the tech-specs table on the product page?
"SR312 Electrical Specification" row

at https://docs.arduino.cc/hardware/braccio-carrier

metanav, 2022-09-02:
Thanks @marqdevx, I found those information except acceleration on the product page. By the way, is there datasheet available for the servo motors, SR312 and SR418D? Also, I could not find STL or DAE files for the Braccio++ arm parts (not the carrier board)

marqdevx, 2022-09-02:
I could not find the acceleration 🤔
About the 3D files, let me ask.

marqdevx, 2022-09-05:
To get the acceleration:
```
Force  = Torque / lever length
Acceleration = Force / mass (depending on the target)
```

As it is not a fixed value we cannot add it to the tech-specs table

closed 2022-09-06 by marqdevx