arduino/docs-content #528: Possible Typo In Documentation For Digital Pins

bluebandit21, 2022-08-29:
The documentation for digital pins has a special section regarding pin 13 behaving oddly when used as an input due to the attached LED and resistor.

> NOTE: Digital pin 13 is harder to use as a digital input than the other digital pins because it has an LED and resistor attached to it that's soldered to the board on most boards. If you enable its internal 20k pull-up resistor, it will hang at around 1.7V instead of the expected 5V because the onboard LED and series resistor pull the voltage level down, meaning it always returns LOW. If you must use pin 13 as a digital input, set its pinMode() to INPUT and use an external pull down resistor.

I'm confused by the following line which doesn't make sense to me.
> If you must use pin 13 as a digital input, set its pinMode() to INPUT and use an external pull down resistor.

If the pin mode was set to INPUT and nothing was driving it, it seems like the path to ground through the attached LED and resistor would lead to the pin consistently reading as low anyway, making the addition of a pull down resistor pointless (quick testing on an UNO R3 I have matches this expected behavior).

Was the documentation meant to instead say something along the lines of this instead?
> If you must use pin 13 as a digital input with a pullup, set its pinMode() to INPUT_PULLUP and use an external pull up resistor.

marqdevx, 2022-08-31:
By adding an external pull down resistor, it means that you can play with the voltage of that pull down, for example if the LED requires 3V, and you want to have 5V "available" you should have 8V on the external pull down.

(The numbers are not real, it depends on the board's voltage and the specific board's LED)

marqdevx, 2022-08-31:
Could you share the tutorial link? 👀 😄

bluebandit21, 2022-08-31:
Of course!

https://www.arduino.cc/en/Tutorial/Foundations/DigitalPins

https://github.com/arduino/docs-content/blob/04ae5fcf159bf33b4197cbafa201a21807fb4008/content/learn/02.microcontrollers/01.digital-pins/digital-pins.md

bluebandit21, 2022-08-31:
I thought about this slightly more last night --

I originally proposed changing it to 
> If you must use pin 13 as a digital input with a pullup, set its pinMode() to INPUT_PULLUP and use an external pull up resistor.

but I think it should probably be

> If you must use pin 13 as a digital input with a pullup, set its pinMode() to INPUT and use an external pull up resistor.

instead.

There's no benefit to having two different pullup resistors on the same pin, and if the voltage they're being pulled up to differs you get constant current flow between the two sources.
(I updated the original issue with this small change too)

bluebandit21, 2022-09-16:
Hey, just sending a quick follow-up since it's been a little bit :)

Is this the right place for this issue over, say, the main Arduino repository?

Additionally, would it be helpful for me to go slightly more into detail as to what I think the problem with the current documentation is and what it should changed to be instead?
I can also make a PR with my proposed changes if that would be helpful.

Thanks!

marqdevx, 2022-09-20:
Yes it is the right place to report it 😄, I need to check it out, cause I've just had a quick look, I will get back to you as soon as I can

bluebandit21, 2024-07-08:
Hi, just following up on this again lol

Would it be more helpful if I made an actual pull request with my proposed wording changes?

jcarolinares, 2025-01-22:
@marqdevx any idea about what was the lates status of this case? just to see if content can try to take a quick look