arduino/docs-content #733: Current Sensing

sloopjohnt, 2022-12-16:
It would be good f there was a pointer to a tutorial on using Current Sensing.

marqdevx, 2023-01-23:
Hello @sloopjohnt are you looking for it in an specific device? or a generic tutorial on current sensing?
Thanks

sloopjohnt, 2023-01-24:
@marqdevx I'm looking for better information/advice/learning material for controlling DC motors, specifically 6V TT motors, but that might well change, using the Arduino Motor Shield REV 3. I'm especially looking for material to help me use the current sensing to detect stall conditions (and potentially motor overload/ burnouts) and more subtle loading changes - e.g. a cart starting to push an obstructing object. 

Does that make sense? 

I've got a little prototype working now with the library (but an unofficial shield after burning the original one out), but still haven't managed to code anything which really seems useful to monitor the current sense pins. Just ask if more on the prototype would be useful.