arduino/docs-content #744: preferences.txt switches are not documented

QuadCorei8085, 2022-12-26:
I was browsing through the net for explanation of what switches are possible and how they behave. Found no site where it is described what the switches/options under preferences.txt are doing.

per1234, 2022-12-29:
The existence of the `preferences.txt` file is mentioned here:

https://github.com/arduino/docs-content/blob/main/content/hacking/01.software/Preferences/Preferences.md

but the configuration properties are not documented.

An example of some properties of interest, for things that can only be configured by editing `preferences.txt` (not configurable via the Arduino IDE GUI):

- `editor.auto_close_braces`
- `editor.font`
 `compiler.cache_core`
 - This was previously controlled by the **File > Preferences > Aggressively cache compiled core** option. However, that option has been removed.
 - See: https://github.com/arduino/arduino-cli/issues/464#issue-514590131
- `preferences.readonly`
  - See: https://github.com/arduino/Arduino/pull/7879

The important thing to note is `preferences.txt` is specific to Arduino IDE 1.x. Arduino IDE 2.x does has a completely separate preferences store and does not use `preferences.txt` at all. For the most part, all Arduino IDE 2.x's settings can be configured via the GUI (some in **File > Preferences** and others in the [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings))

QuadCorei8085, 2022-12-29:
So it is undocumented indeed.

Is there at least a complete list with all the properties and swtiches?

per1234, 2022-12-29:
>Is there at least a complete list with all the properties and swtiches?

No. There is no formal official documentation whatsoever of the Arduino IDE 1.x `preferences.txt` properties.

It is unfortunate that was never created. The Arduino IDE 1.x developers added valuable configuration options, but that value was never really delivered to the users because there was no good way to learn about the features.

But the Arduino IDE 1.x project is no longer under active development and Arduino IDE 2.x is the future, so it is questionable whether it is worth the effort to produce such documentation at this point in time.

jcarolinares, 2024-06-06:
Hello @jacobhylen 

As exposed by @per1234 the Arduino 1.X is not longer under active development and therefore I consider this potential documentation is not a priority.

However, the project belongs to the community, and the community if need it can create a Pull Request to expand the documentation of https://github.com/arduino/docs-content/blob/main/content/hacking/01.software/Preferences/Preferences.md to expand that content.

Please take a look and actions if need it, thanks