arduino/docs-content #751: Link to Portenta Datasheet broken

wero1414, 2023-01-03:
When i click to download the datasheet i got a Page not found.

https://docs.arduino.cc/hardware/portenta-h7
[image]

with this URL = https://docs.arduino.cc/static/4cda27067a048e59bc93c31edc5efc1a/ABX00042-ABX00045-ABX00046-datasheet.pdf

akash73, 2023-03-02:
confirmed here. 
links on Portenta H7 lite and Portenta Lite Connected (that are the same as for H7) work. 
Here's the right link https://docs.arduino.cc/static/8b12393ba695296f668d69d469d04d3f/ABX00042-ABX00045-ABX00046-datasheet.pdf

jhansson-ard, 2023-04-20:
The current link on Portenta H7 works, I will close this issue now. Please let us know if there are any more issues! ⭐

closed 2023-04-20 by jhansson-ard