arduino/docs-content #773: Links missing to reference for Stream and Print classes in the Library Style Guide

laserguru, 2023-01-14:
In the body of the [Library Style Guide](https://github.com/arduino/docs-content/blob/main/content/learn/08.contributions/01.arduino-library-style-guide/arduino-library-style-guide.md) users are asked to, quote:

> **Use the established core libraries and styles.**
> [...]
> * Use the `Stream` and `Print` classes when dealing with byte streams.

However there aren't any links to the reference for those classes. Moreover I don't think there is documentation for them yet in these new docs. There's documentation for the `Stream` class [in the Reference](https://www.arduino.cc/reference/en/language/functions/communication/stream/), but it seems there isn't any for the `Print` class (at least judging by a quick search)

per1234, 2023-01-14:
>it seems there isn't any for the `Print` class (at least judging by a quick search)

The need to add general purpose `Print` class documentation to the Arduino Language Reference is tracked here:

https://github.com/arduino/reference-en/issues/786

marqdevx, 2023-01-23:
Can I close this issue as this is handled externally?

per1234, 2023-01-23:
Hi @marqdevx. We can break the resolution of this request into individual tasks. One of those tasks is for work in the `arduino/reference-en` repository, so that specific task must be handled externally.

However, the other two tasks are for work in this repository. So no it should not be closed.

I'll list the tasks:

## Add `Stream` class documentation link to Library Style Guide 

The task is to change the content of the "[**Arduino Style Guide for Creating Libraries**](https://github.com/arduino/docs-content/blob/main/content/learn/08.contributions/01.arduino-library-style-guide/arduino-library-style-guide.md)", which is hosted in this repository.

The `Stream` class documentation is here:

https://www.arduino.cc/reference/en/language/functions/communication/stream/

So this task is not blocked by any external work and can be done at any time.

## Write `Print` class documentation

The task is to write and add the missing documentation content in the `arduino/reference-en` repository.

That task is tracked here: https://github.com/arduino/reference-en/issues/786

## Add `Print` class documentation link to tutorial

The task is to change the content of the "[**Arduino Style Guide for Creating Libraries**](https://github.com/arduino/docs-content/blob/main/content/learn/08.contributions/01.arduino-library-style-guide/arduino-library-style-guide.md)", which is hosted in this repository.

The `Print` class documentation does not currently exist so this task is blocked by https://github.com/arduino/reference-en/issues/786

igmtz, 2023-04-05:
Hello @per1234 can I help in some of this tasks? Is there any example of how the links must be added to the Library Style Guide Page?