arduino/docs-content #845: Portenta H7 Issue

Bilalgp, 2023-02-15:
I am having an issue having pin A4 read and pin D10 digital output simultaneously. They appear to be connected in some way. If I run the following code, D10's output depends on the incoming analog signal. 

I tried multiple Portentas and all are suffering from the same problem. I am wondering if this is a hardware or software issue. I am integrating this Portenta on a fixed circuit board, so changing the utilization of the pins is difficult. 

```cpp
int max_A0;
int max_A1;
int max_A2;

int sensorValue1, sensorValue2;

void setup() {
  Serial.begin(9600);

  //pinMode(15, INPUT);
  //pinMode(16, INPUT);
  //pinMode(18, INPUT);
  //pinMode(19, INPUT);
  //pinMode(20, INPUT);
  pinMode(10, OUTPUT);
  pinMode(0, OUTPUT);
  pinMode(4, OUTPUT);
  delay(1000);
  //digitalWrite(D10, OUTPUT);  // sets the digital power ON (D10)
  //digitalWrite(clock1, OUTPUT);  // sets the digital power ON (D10)
  digitalWrite(0, HIGH);  // sets the digital pin D4 off
  //digitalWrite(10, HIGH);  // sets the digital power ON (D10)
  //digitalWrite(4, HIGH);
  delay(1000);
  digitalWrite(10, HIGH);
}

int i = 0;
void loop() {
  //digitalWrite(10, HIGH);
  delay(10);
  //digitalWrite(10, HIGH);
  //pinMode(19, INPUT);
  //delay(10000);
  //pinMode(19, INPUT);
  max_A0 = analogRead(20);
  delay(10);
  sensorValue1 = analogRead(15);
  delay(10);
  sensorValue2 = analogRead(16);
  delay(10);

  max_A2 = analogRead(18);
  delay(10);  
  //pinMode(19, INPUT);
  max_A1 = analogRead(A4);
  delay(10);
  //pinMode(10,OUTPUT);
  Serial.println(max_A1);
  //delay(1000);

  //digitalWrite(10, HIGH);

  //if(i < 5) {
  //  pinMode(10,OUTPUT);

  //pinMode(10,INPUT);
  // }
  i++;
  //int max_A1 = analogRead(19);
  // Serial.println(max_A1);
}
```

Bilalgp, 2023-02-15:
I think I understand. Thank you. I guess it's a bit confusing because this would mean, PC2 is a GPIO but according to the[ portenta H7 schematic ](https://github.com/arduino/ArduinoCore-mbed/issues/488#issuecomment-1142688565) A4 does not function as an digital pin...but it is connected to D10.

jcarolinares, 2023-11-14:
Issue seems to be resolved, thank you all!

closed 2023-11-14 by jcarolinares