arduino/docs-content #1096: Modbus sample code does not work

robertoperuzzo, 2023-06-17:
Thank you very mutch for the awesome [doc article about Modbus](https://github.com/arduino/docs-content/blob/main/content/learn/05.communication/07.modbus/modbus.md).

I'm testing the proposed code snippet (copied down here)

```
float readVoltage() {
  float volt  = 0.;
  // Send reading request over RS485.
  if (!ModbusRTUClient.requestFrom(SLAVE_ADDRESS, INPUT_REGISTERS, 30017, 2)) {
    // Error handling
    Serial.print("- Failed to read the voltage!");
    Serial.println(ModbusRTUClient.lastError());
  } else {
    /*
     * Response handler
     */
    // Read word1 from buffer
    uint16_t word1 = ModbusRTUClient.read();
    // Read word2 from buffer
    uint16_t word2 = ModbusRTUClient.read();
    // Join word1 and word2 to retrieve voltage value in millivolts
    uint32_t millivolt = word1 << 16 | word2;
    // Convert to volts
    volt = millivolt/1000.0;
  }

  return volt;
}
``` 

But I receive this error
```
- Failed to read the voltage!Illegal data address
```

So, reading the [Finder 7M.24 Modbus communication guide](https://robertoperuzzo.it/sites/default/files/2023-06/Modbus-7M24-7M38_v2_30062021.pdf) I changed the third parameter in requestFrom() with the hex 0x6B (= decimal 107) and the meter respond, but with the **weird number 4278192.50**.
Watching the Finder, **it is measuring ~237,80V**

Anyone can help me to understand what I'm doing wrong?

Thank you.

robertoperuzzo, 2023-06-19:
@akash73 probably you may help me

jcarolinares, 2023-11-14:
@TaddyHC any change this is related to the work we are developing? 

@robertoperuzzo please take a look too at : [https://docs.arduino.cc/tutorials/opta/energy-management-application-note](https://docs.arduino.cc/tutorials/opta/energy-management-application-note). The example it is more complex (being part of an application note) but we know that it works with the Finder 7M.24 (checked last month) so can be part of the solution.

TaddyHC, 2023-11-14:
Hello @jcarolinares, upon checking the documentation, I've identified a typo concerning the third argument of the requestFrom() method within the code. We've been using 30017 as an argument inside the code instead of 30107, as stated in the tutorial. So, we'll have to patch the code of the documentation. As for the weird number, I may have noticed an issue with the data encoding.

To be sure, we would have to test it again by checking the data encoding for the T5 data type, which uses a dedicated encoding:

float convert_t5(uint32_t n)
{
    uint32_t s = (n & 0x80000000) >> 31;
    int32_t e = (n & 0x7F000000) >> 24;
    if (s == 1) {
        e = e - 0x80;
    }
    uint32_t m = n & 0x00FFFFFF;
    return (float)m * pow(10, e);
}

jcarolinares, 2024-03-26:
Looks like @robertoperuzzo didn't reply after some months so hopefully the problem is already solved for him.

In any of the cases if you experiment more problems please do not hesitate to ask us for help at: https://www.arduino.cc/en/contact-us/

Thank you!

closed 2024-03-26 by jcarolinares