arduino/docs-content #1148: Duplicated PLC/MicroPython tutorials on https://docs.arduino.cc/tutorials/

aliphys, 2023-07-06:
## Current State
In the https://docs.arduino.cc/tutorials/ page, there are duplicated tutorials for PLC (red) and MicroPython (blue).
[image]

In the case MicroPython, the links are:
- https://docs.arduino.cc/tutorials/portenta-h7/micropython-installation
- https://docs.arduino.cc/tutorials/portenta-h7-lite/micropython-installation
- https://docs.arduino.cc/tutorials/portenta-h7-lite-connected/micropython-installation
- https://docs.arduino.cc/tutorials/nano-rp2040-connect/micropython-installation
- https://docs.arduino.cc/tutorials/nano-33-ble/micropython-installation

The content on the pages look all the same. Only difference is the product name at the top of the page.

[image]

## Expected State
- Duplicate tutorials which correspond to multiple products are consolidated into one entry. One possible proposal: 
[image]

jhansson-ard, 2023-07-06:
Thank you for your report! Our team is on it and will hopefully have a solution soon ⭐

jcarolinares, 2025-01-22:
outdated, closing

closed 2025-01-22 by jcarolinares