arduino/docs-content #1156: Getting started for Arduino uno r3 still uses Arduino IDE 1.x.x

V205Github, 2023-07-09:
Getting started for Arduino uno r3 still uses Arduino IDE 1.x.x
Should this be changed to use Arduino 2.x.x since this is the newest release?

karlsoderby, 2023-09-08:
Hi @V205Arduino . Thanks for reporting this, and it is indeed true for a number of legacy boards that were released prior to the release of IDE 2. 

We are working on a general solution for this type of documentation, but for now I can't tell you when that will be released!

V205Arduino, 2023-09-08:
Ok thanks!  Anything I can contribute to?

karlsoderby, 2023-09-28:
Hi again @V205Arduino and sorry for the late reply. We have updated the docs, so that you are now linked to a generic installation guide for **any** Arduino core.
- This guide shows at the top how to install a core (very nicely via the built-in board manager in the IDE 2)
- It has a specific section (in this case, it is AVR) for every core, and which board is linked to it.

Thank you! I am closing this issue but feel free to open it again in case there's still some open issues. Have a great day!

closed 2023-09-28 by karlsoderby

V205Arduino, 2023-09-28:
Ok thanks!