arduino/docs-content #1274: bitRead and bitWrite documentation ambiguity

mjbmikeb2, 2023-08-24:
Bearing in mind the target audience is beginners, bitRead says "Reads a bit of a number." 
Is that all signed and unsigned numbers of any size?

bitWrite says, "Writes a bit of a numeric variable"
Again, is that all signed and unsigned numbers of any size?

According to this thread
https://forum.arduino.cc/t/bitread-and-bitwrite-for-numbers-bigger-than-a-byte/1160471
bitWrite doesn't work for 64 bit numbers.

I don't know if this is a bug or a feature. Either way can the documentation please be updated to say what datatypes are currently supported.

Thanks.

jhansson-ard, 2023-08-31:
Hey @mjbmikeb2 and thank you for your comments! 

Is there any specific documentation page you are referring to, or is this more of an overall observation?

mjbmikeb2, 2023-09-01:
The pages are 
https://www.arduino.cc/reference/en/language/functions/bits-and-bytes/bitread/
https://www.arduino.cc/reference/en/language/functions/bits-and-bytes/bitwrite/

jhansson-ard, 2023-09-01:
Thank you! 

If you haven't already, could you add your comment in the issue section of this repository instead? 
https://github.com/arduino/reference-en

That's where we handle the Arduino Reference!

mjbmikeb2, 2023-09-01:
Moving this as requested to https://github.com/arduino/reference-en/issues/924
Closing this issue.

closed 2023-09-01 by mjbmikeb2