arduino/docs-content #1322: Error in cheat sheet documentation for WiFi

Grumpy-Mike, 2023-09-10:
Under the USB Bridge section. The document says:-
Software - By pulling D40 to HIGH you will close the circuit that controls which MCU is connected to USB. While D40 is HIGH, the RA4M1 is connected to the USB Serial port, and while D40 is LOW the ESP32 is connected, like the default configuration. You can do this by including the following code in
void setup()
------------
Observation using an Oscilloscope on the select line shows that using D40 does not change the select line. Further observation shows that pin 21 does change the data select line. In addition the state of this line is the inverse of the information shown that section quoted above. So that section of the documentation should say:-
-----------
Software - By pulling D21 to LOW you will close the circuit that controls which MCU is connected to USB. While D21 is LOW, the RA4M1 is connected to the USB Serial port, and while D21 is HIGH the ESP32 is connected. You can do this by including the following code in void setup()

> pinMode(21, OUTPUT);
> digitalWrite(21, LOW);
>

delta-G, 2023-09-10:
My test [here](https://forum.arduino.cc/t/are-arduino-not-intrested-in-getting-feedback/1167164/4) shows this to be right about the pin but backwards on the logic.  I see the ESP with 21 HIGH and the RA4 with 21 LOW.

Grumpy-Mike, 2023-09-11:
" I see the ESP with 21 HIGH and the RA4 with 21 LOW."

Yes, so do I, why does this make my logic backwards?

facchinm, 2023-09-16:
Relevant code:
https://github.com/arduino/ArduinoCore-renesas/blob/590ab1535be7f1028706c9f9dd77c01206a8c1dc/variants/UNOWIFIR4/variant.cpp#L94-L101

closed 2023-09-29 by jacobhylen

fumikrobot, 2024-02-15:
after i set 21 to HIGH, the board can't detected by computer. Now, it can't communicate with computer for anything, even Arduino IDE. How to "wake it up"?

Grumpy-Mike, 2024-02-15:
I can get the MIDI library to work on a Minima, no problem. When I try to use it on the WI-FI it crashes and refuses to load on subsequent downloads. In fact it never gets to the upload part citing an error with the serial part not being there. So the boards needs to be somehow recovered before it will work again.

This can be done in two ways

A double tap on the reset button

A power cycle of the board.

This depends on the state of that USB switch data select.
The problem with 1) is that after the board is recovered then there will not be a serial port created of the correct baud rate to attach to Hairless and will show hairless errors that clearly indicate the baud rate is wrong.

This doesn't happen with method two.

I am very aware of the schematic and I used it to probe the board on an oscilloscope and confirm that magic number 40 did not switch the select line of the USB but magic number 21 did switch it. This allowed the recovery of the board using a power cycle which did produce a serial port with the correct data rate.

Grumpy-Mike, 2024-02-15:
Also Read
https://forum.arduino.cc/t/midi-library-works-on-minima-but-not-on-wifi/1165783 <https://forum.arduino.cc/t/midi-library-works-on-minima-but-not-on-wifi/1165783>

https://forum.arduino.cc/t/uno-r4-wifi-fails-at-ble-begin/1172455 <https://forum.arduino.cc/t/uno-r4-wifi-fails-at-ble-begin/1172455>

fumikrobot, 2024-02-15:
thanks for your quick reply. I solve it by myself: short the pin 21 (i guess so, it is at back of the board (USB-RA4M1)) to GND, then the computer can detect the board at COM3 (also same at Arduino IDE). Then, upload another code to release Pin 21. After that, the board become normal again.

delta-G, 2024-02-26:
Setting pin 21 HIGH will cause the board to be no longer connected to the computer because you have switched the USB to the RA4M1 chip and it doesn't have any USB code running.  There is code you can add to enable the USB_Serial on the RA4M1 and then you can communicate with it like a MINIMA.