arduino/docs-content #1432: Problem with Saving MQTT Payload Data to CSV or TXT File from Portenta X8

marimelhanafy, 2023-10-23:
Hello everyone,

Following tutorial no. 12 (https://docs.arduino.cc/tutorials/portenta-x8/multi-protocol-gateway), I've encountered an issue with saving data from the Python code to a csv or even a txt file.
I'm using an ESP32, which sends the data to the Portenta X8 that's connected to the Max carrier. When I run the code, the data is received and printed, but it's not saved to the file. Could the problem be related to the Arduino and Linux layers?

Here's the data being successfully sent from the ESP32 to the Portenta X8:
[image]

Here is the modification I made to the part of the code:
[image]

I tried using f.write(str(float(decoded_mqtt_payload))) and f.write('hello'). However, both attempts did not work, and the resulting file is totally empty.

Thanks in advance.

marimelhanafy, 2023-10-23:
@manchoz @facchinm @per1234 @cturvey @dcuartielles @pillo79, I am in need of help here. Could you please assist me? 
Sorry for any inconvenience!

per1234, 2023-10-23:
@marimelhanafy thanks for taking the time to report the potential problem with the tutorial.

However, this is not an appropriate place to request project assistance help and spamming a bunch of people with mentions is very rude. The forum is the appropriate place to request assistance. I see you have already created a topic there:

https://forum.arduino.cc/t/problem-with-saving-mqtt-payload-data-to-csv-or-txt-file-from-portenta-x8/1181185

jcarolinares, 2023-11-14:
Being the issue already solved at https://forum.arduino.cc/t/problem-with-saving-mqtt-payload-data-to-csv-or-txt-file-from-portenta-x8/1181185 I proceed to close the ticket. Thank you all!

closed 2023-11-14 by jcarolinares