arduino/docs-content #1465: cdc port dirver install fails (Win7, Uno R4 minima)

JwX000, 2023-11-03:
I have problem connecting Arduino Uno R4 Minima to Windows 7 host (IDE 1.8.10 and 1.8.19) - after connecting USB, CDC port driver fails to install ("failure of driver installation" window appears). 
Code upload works, but anything related to serial port doesn't. 
Connecting to Win10 machine is OK - everything installs correctly. Unfortunately, I don't have another Win7 machine to check if this problem can be replicated elsewhere. 
System log contains information that installation of driver:
_FileRepository\renesas.inf_amd64_neutral_b200538e73fc29f0\renesas.inf_ 
for device
_USB\VID_2341&PID_0069&MI_00\7&28FC284E&0&0000_ 
returned _0xe0000219_

per1234, 2023-11-03:
> I would mark Win7 as unsupported

Thanks for the clarification @facchinm. This means it is a documentation task so I have transferred the issue to the `arduino/docs-content` repository.

JwX000, 2023-11-03:
Hello!
It may be some local configuration problem - after opening zadig there was already another mapping for USB\VID_2341&PID_0069&MI_02 (into WinUSB driver) - didn't notice this at first, remapped MI_00, got detected but unusable port (serial monitor cannot set parameters), then found MI_02, remapped it and it seems to work (using AnalogReadSerial example)