arduino/docs-content #1491: Misleading current consumption in Nano ESP32 documentation

PTSolns, 2023-11-10:
On page 2 of the [Arduino Nano ESP32 documentation](https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf), under "Low Power" subheading it states that the deep sleep current consumption is 7uA (first bullet point). In the context with which this is written, the reader may easily assume that the entire Nano board is meant by that and it consumes only 7uA relating to the entire board (during deep sleep). When in reality it is *only* the ESP32 chip that consumes 7uA during deep sleep. The power management components on the board consume current also during deep sleep, which exceeds 7uA.

The entire Nano ESP32 board consumes much more than that (based on our measurements) and Arduino Support has confirmed that current consumption tests of the board during deep sleep have not been done. Therefore, given this, I suggest that the documentation is updated to either A) update that the 7uA current consumption is only for the ESP32 chip, and not the entire Nano board, or B) take the bullet point referring to the 7uA out entirely.

karlsoderby, 2023-11-23:
Hi @PTSolns , you are correct. This information is specific to the ESP32-S3 MCU, and not to the Nano ESP32. I will clarify this in the data sheet. Thanks for the submission.

karlsoderby, 2023-11-24:
Hi @PTSolns , changes have been made in the data sheet, see changes here: https://github.com/arduino/docs-content/pull/1530

feel free to reopen this issue if the issue remains!

closed 2023-11-24 by karlsoderby