arduino/docs-content #1525: Assistance Required with Portenta X8 Analog Pin Reading and Sampling Rate Query

marimelhanafy, 2023-11-22:
I am currently utilizing the Portenta X8 in conjunction with the Max carrier. My goal is to read the voltage using the J5 HDC Stacker Connector. To do this, I am attempting to read from the analog pin A0 using Python in a Linux environment. So, I am using the Linux sysfs interface as stated in this document (Page 2): `https://docs.arduino.cc/resources/pinouts/ABX00043-full-pinout.pdf`.

Here is the code (Python):
```
ser = serial.Serial()
ser.baudrate = 19200
ser.port = '/sys/bus/iio/devices/iio:device0/in_voltage0_raw'
ser.bytesize = 8
ser.parity = 'N'
ser.stopbits = 2
ser.timeout = 1
logging.debug("Configured serial port with:\n\r%s" % str(ser))
logging.debug("Opening serial port")
```

I have also included the path to the analog pin in the docker-compose.yml file's volumes section, as shown below:
```
volumes:
  - '.../sys/bus/iio/devices/iio:device0/in_voltage0_raw'
```

But I am getting this error: "Error response from daemon: invalid volume specification: '/sys/bus/iio/devices/iio:device0/in_voltage0_raw:rw': invalid mount config for type "bind": invalid mount path: 'device0/in_voltage0_raw' mount path must be absolute."

I would appreciate any assistance you can provide in resolving this issue or provide me with a turorial how to use the pins in this document (Page 2): `https://docs.arduino.cc/resources/pinouts/ABX00043-full-pinout.pdf`

Additionally, I would like to ask about the sampling rate of the Portenta X8. Could you provide me with this information?

Thanks in advance.

marimelhanafy, 2023-11-27:
I have updated the devices section in my docker-compose.yml file, and commented out previous attempts made in the volumes section, as shown below:
```
devices:
  - '/dev/ttymxc3'
  - '/dev/gpiochip5'
  # - './mydevices:/sys/bus/iio/devices:rw'
  # - '/var/rootdirs/home/fio/sys/bus/iio/devices/iio:device0/in_voltage0_raw:rw'
  # - '/sys/bus/iio/devices:rw'
  # - '/sys/bus/iio/devices/iio:device0/in_voltage0_raw:rw'
  # - '/sys/bus/iio/devices'
  - '/sys/bus/iio/devices/iio\:device0/dev'
```
Despite trying all the commented options, I still encounter the following error: 
[image]

Please note that Docker interprets the colon as a delimiter between the host path and the container path in the volume specification. If the file or folder name in the path contains a colon, it might not be correctly interpreted. This issue probably arises from Docker's interpretation, not from Python code. I have searched for a solution using the --mount flag, but it works with Docker volumes and not for devices (when using the --device flag). This appears to be a known limitation in Docker.

So, my questions are: Could you suggest a solution for this? And, is it possible to change the name of the folder 'iio:device0'?

jcarolinares, 2023-12-19:
@mcmchris can we help the user with this problem please? any idea?

mcmchris, 2023-12-19:
@marimelhanafy 

## About Sampling Rate:
- The **A0** pin of the Portenta X8 is routed to the **PF11** pin of the STM32H747AII MCU. 
- The MCU counts with three groups of ADC channels ( Direct channels, Fast channels and Slow channels).
- The **PF11** is part of the _Fast channels_.
- The max sampling rate of the channels depends on the configured ADC resolution and ADC frequency.
- See the following table:
[image]
- You can reach up to 7.10 MSps with the _Fast channels_.

On the [MCU datasheet](https://www.st.com/resource/en/datasheet/stm32h747ai.pdf) (page 185), there is a table with the detailed information.

## About reading the A0 ADC:
From the Portenta X8 shell you can read the A0 pin ADC value with:

`cat /sys/bus/iio/devices/iio\:device0/in_voltage0_raw`

[image]

From a Python script you can do it this way:

```python
def read_adc_value(adc_pin):
  try:
    with open(f'/sys/bus/iio/devices/iio:device0/in_voltage{adc_pin}_raw', 'r') as file:
        return int(file.read().strip())
  except FileNotFoundError:
    print(f"ADC pin {adc_pin} not found!")
    return None
```

Here is a full example of a script that let you enter manually the channel to be sampled:

```python
def read_adc_value(adc_pin):
  try:
    with open(f'/sys/bus/iio/devices/iio:device0/in_voltage{adc_pin}_raw', 'r') as file:
        return int(file.read().strip())
  except FileNotFoundError:
    print(f"ADC pin {adc_pin} not found!")
    return None

if __name__ == "__main__":
  adc_pin = input("Enter ADC pin number: ")
  value = read_adc_value(adc_pin)

  if value is not None:
    print(f"Value from ADC pin {adc_pin}: {value}")

    # Mapping between 0-3.3 V
    new_value = (float) (value/65535)*3.3
    print(f"Value mapped between 0-3.3 V: {new_value}")
```
[image]

cc: @jcarolinares

marimelhanafy, 2023-12-21:
Could you please elaborate more about this line:
```
    # Mapping between 0-3.3 V
    new_value = (float) (value/65535)*3.3
```
Why did you chose this value (65535) for the division?

marimelhanafy, 2023-12-21:
> @marimelhanafy
>
> ## About Sampling Rate:
> * The **A0** pin of the Portenta X8 is routed to the **PF11** pin of the STM32H747AII MCU.
> * The MCU counts with three groups of ADC channels ( Direct [...]

Thank you so much for this helpful reply!
Could you please elaborate more about this line:
```
    # Mapping between 0-3.3 V
    new_value = (float) (value/65535)*3.3
```
Why did you chose this value (65535) for the division?

mcmchris, 2023-12-21:
@marimelhanafy The Portenta X8 ADC resolution is set to 16 bits by default. This means that it will split the whole 3.3v range on 2<sup>16</sup> pieces = 65536 - 1 = 65535, we subtract 1 to obviate the "0". 

Dividing the ADC value by 65535 will return a fractional parameter from 0 to 1 (0% to 100%) that multiplied by 3.3 will return a voltage related value.

marimelhanafy, 2023-12-24:
> @marimelhanafy The Portenta X8 ADC resolution is set to 16 bits by default. This means that it will split the whole 3.3v range on 216 pieces = 65536 - 1 = 65535, we subtract 1 to obviate the "0".
> 
> Dividing the ADC value by 65535 will return a fractional parameter from 0 to 1 (0% to 100%) that multiplied by 3.3 will return a voltage related value.

Thank you so much! Everything is clear now.

closed 2023-12-24 by marimelhanafy