arduino/docs-content #1615: Outdated graphic for Arduino Cloud API key management

per1234, 2023-12-16:
The Arduino Cloud website was [recently redesigned](https://blog.arduino.cc/2023/11/15/announcing-a-better-arduino-cloud-together/). Before the redesign, API keys were managed by selecting "API keys" from the navigation panel on the Arduino Cloud home page.

That item was removed from the navigation panel and there is now a dedicated page for managing API keys. This means the graphic illustrating the use of the old interface is now outdated and potentially misleading to the user:

https://github.com/arduino/docs-content/blob/aa7c281ca5a298c7489f462dbef02a2d880c1bd7/content/arduino-cloud/07.api/02.arduino-iot-api/assets/api-keys.png

[image]

The new interface looks like this:

[image]

The update for the accompanying text is already submitted at https://github.com/arduino/docs-content/pull/1614

### Additional Context

Originally reported at https://forum.arduino.cc/t/api-key-generation-page-not-available-on-home-screen/1200517

g-castronovo, 2024-07-18:
[This tutorial](https://docs.arduino.cc/arduino-cloud/api/arduino-iot-api/)
still shows old graphic ;;

jcarolinares, 2025-06-16:
This is part of the whole documentation revamp regarding this tipic we are doing, being already part our our internal ticket system.

closed

cc @jhansson-ard

closed 2025-06-16 by jcarolinares