arduino/docs-content #1617: Analog/Calibrate Example Incorrect circuit connetion

Koler93, 2023-12-17:
[image]
 A0(Yellow line) should be connected to i10 and GND (black line) port of arduino to i13 on breadboard.

jhansson-ard, 2023-12-27:
Hello @Koler93! Please include a link to the tutorial where you found this issue, so that we can take a look and work on a solution.

Thank you! ⭐

Koler93, 2023-12-28:
https://docs.arduino.cc/built-in-examples/analog/Calibration here is the link to tutorial, I forgot to inlcude when i was opening issue.

pedromsousalima, 2024-08-01:
https://github.com/arduino/docs-content/pull/1747 the circuit has been fixed on this PR.
Thank you for the feedback!

closed 2024-08-01 by pedromsousalima