arduino/docs-content #1752: Document Arduino HAT Carrier Pins to Arduino IDE Pins

hybotics, 2024-02-11:
This is documentation that is very much needed.

There does not seem to be any documentation that maps the pins on the 40-Pin connector to Arduino IDE Pins. Without this documentation, it is impossible to use the HAT Carrier in any project. Why are there no other compatible Raspberry Pi HATs that are usable on the HAT Carrier?

I am building a robot that uses the Arduino Portenta C33 on the Arduino HAT Carrier. I need to know how to access all the pins on the Raspberry Pi compatible 40-pin connector. I will be using Raspberry Pi HATs that are apparently not supported.

jcarolinares, 2024-02-12:
Hello!

Indeed, we created a Pinout Table for Hardware and also for Software.

Due to some technical limitations that information is available at this moment inside the full pinout:
 https://docs.arduino.cc/resources/pinouts/ASX00049-full-pinout.pdf

Please go to page 6

[image]

During this week the Pinout Table will be available as an independent document at:
 https://docs.arduino.cc/hardware/portenta-hat-carrier/

There you will find what are you looking for. In case you need more help please don't hesitate to contact us :

https://www.arduino.cc/en/contact-us/

hybotics, 2024-02-15:
Can I use the pin names on Arduino HAT Carrier by the names in the **_Portenta HD Standard Pin_** column in my new header file?

My goal is to easily relate the functions of the Raspberry Pi to the Arduino Portenta pins.

jcarolinares, 2024-02-17:
As we talk about it on Discord you must use column named as "Portenta C33" from page 6

[image]

From the documentation point of view, I consider the issues as solved. But please do not hesitate contact Arduino Support or use other ways in case you still have problems from the technical point of view!

Happy coding!

closed 2024-02-17 by jcarolinares

hybotics, 2024-02-17:
I have still not been able to access any of the peripherals on the Raspberry Pi connector of the Arduino HAT Carrier and I do not know why. As far as I can tell, I have everything correct, but I am not 100% sure. I am using the pins according to the documentation for the HAT Carrier, but nothing happens when I access them. I have two switches connected from GROUND through a 220 ohm resistor, which is connected to I/O pin 53, with the same thing for pin 54.

stackovercode, 2024-12-04:
This is still an open issue, I can confirm that it is still not possible to connect and use the portenta hat carrier together with the Portenta H7. No Arduino Library or manual exist to complete utilize the pin layout.