arduino/docs-content #1839: Confusing Sentence in Powering Alternatives for Arduino Boards

Batvos, 2024-03-25:
Hey,

I noticed some confusion in the documentation regarding power pins usage on Arduino.
In the section about using the 3V3 and 5V pins as power inputs on https://docs.arduino.cc/learn/electronics/power-pins/  there is a sentence that might be misleading: 

> "Although 3V3 and 5V pins can be used as power inputs, **it is not recommended** if no power supply is connected through the USB port, the barrel jack connector, or the VIN pin."

This sounds to me that USB port, the barrel jack connector, or the VIN pin must be connected simultaneously when using 3V3 and 5V pins as power inputs.
I suggest to rephrasing it: "While it's possible to use the 3V3 and 5V pins as power inputs, it's not advisable to do so when other power sources are available".

Also reading further: 

> "Suppose the voltage in the voltage regulator output pin becomes higher than the input voltage of the voltage regulator. In that case, a large current may flow into the voltage regulator from its output pin to its input pin."

This seems relevant only when Arduino is powered by USB port, the barrel jack connector, or the VIN pin. Is it correct? If so, I suggest replacing it with something like: “it’s important not to exceed 5V and 3V3 voltage on the corresponding pins, as they are not regulated by voltage regulator.”

jcarolinares, 2024-03-26:
Hello @Batvos , first of all thank you for opening an issue.

I agree with the first point of your problem and your suggestion, it is up to the content creators to use it or modify it.

Second one is correct but not in accordance with the tone of the rest of the page, so I suggest to rephrase it internally as Arduino keeping your original message

@canchebagur please help in this issue and takes actions in accordance/approval with @karlsoderby as responsible of this part if need it, thanks @Batvos ! :)

jcarolinares, 2024-06-06:
Added  @jacobhylen not being karl available, @canchebagur please help @jacobhylen as mentioned