arduino/docs-content #1876: Images not being displayed on official website

Lawrence4U, 2024-04-14:
None of the images from the docs are displaying on the website view. The markdown inside the github repo works just fine.
https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoToBreadboard/ 
[image]

I have checked only a couple of pages and on both no images where displayed. This happens both on chrome and edge regardless of adblocks etc.

jhansson-ard, 2024-04-15:
Hello @Lawrence4U and thank you for your opened issue! 

We are aware of the problem, and we're working on a solution. This will be solved as swiftly as we can ⭐

jhansson-ard, 2024-05-31:
This is now fixed @Lawrence4U, thanks for your issue report ⭐

closed 2024-05-31 by jhansson-ard