arduino/docs-content #1968: [ACELL01][ACELL07] Portenta X8/C33/H7 compatibility with Portenta Mid Carrier not clearly communicated

aliphys, 2024-05-21:
There are a few inaccuracies in the Portenta Mid Carrier Docs page
[image]
- Technically, all interfaces are accessible to a User who wants to hack at it (yay open source!). Yet, we do have a specific OoB experience that we provide support for (see Details)
- There are some limitations in what we can support with the STM32 included inside the Portenta X8, since the peripherals are shared with the i.MX core.
- The GNSS support is not part of the Portenta Mid Carrier, nor is it provided by both version of the Pro 4G module.

---
# Details
The Portenta Mid Carrier is indeed compatible with the Portenta boards - Portenta C33 ([ABX00074](https://docs.arduino.cc/hardware/portenta-c33)), Portenta H7 ([ABX00042](https://docs.arduino.cc/hardware/portenta-h7) and variants) and Portenta X8 ([ABX00049](https://docs.arduino.cc/hardware/portenta-x8)). The final customer solution will likely involve the Pro 4G EMEA (SKU: [TPX00201](https://store.arduino.cc/products/4g-module-emea)) or GNSS (SKU: [TPX00200](https://store.arduino.cc/products/4g-module-global)) PCIE boards.

Network connectivity is handled differently between the Portenta C33/H7 ([arduino-libraries/Arduino_Cellular/](https://github.com/arduino-libraries/Arduino_Cellular)) and the Portenta X8 (Yocto Linux). One is with a Arduino library, the other is via the Linux stack. 

Here is the information in a table format:

| Portenta Board | Carrier Board       | Pro 4G Module | Cellular Support                |
|----------------|---------------------|---------------|------------------------|
| Portenta C33/H7| Portenta Mid Carrier| EMEA (No GPS)/GNSS (with GPS)           | Arduino_Cellular library |
| Portenta X8    | Portenta Mid Carrier| EMEA (No GPS)/GNSS (with GPS)           |  Yocto Linux             |

This difference is explained inside the Portenta X8 User Manual
https://github.com/arduino/docs-content/blob/42e01e0f82947b8edd85b3450bea4e1de996a6ca/content/hardware/04.pro/boards/portenta-x8/tutorials/01.user-manual/content.md?plain=1#L49-L63

Note that the User Manual conflicts with the statement made here
https://github.com/arduino/docs-content/blob/42e01e0f82947b8edd85b3450bea4e1de996a6ca/content/hardware/04.pro/carriers/portenta-mid-carrier/features.md?plain=1#L18

aliphys, 2024-05-21:
Related https://github.com/arduino/arduino-examples/issues/74

jcarolinares, 2024-06-06:
@TaddyHC do me a favor, please follow the analyses done by @aliphys and send me internal conclusions and potential actions if need it, thanks!

jcarolinares, 2024-12-02:
Task closed in accordance with the original creator of the issue. We are already fixing the problems in different Jira Task internally.

closed 2024-12-02 by jcarolinares