arduino/docs-content #1971: [ACELL01] Inaccessible libraries mentioned in Portenta Mid Carrier User Manual

aliphys, 2024-05-22:
In the Portenta Mid Carrier User Manual we mention three libraries - `Arduino_4GModem`, `Arduino_4G_Module` and `ArduinoProModem` - none of which are accessible via the library manager nor are listed in the [library registery](https://github.com/aliphys/library-registry/blob/7fbd33020f8d49249b4d0ce08d0b3163861b0907/repositories.txt). 

[image]

https://github.com/arduino/docs-content/blob/42e01e0f82947b8edd85b3450bea4e1de996a6ca/content/hardware/04.pro/carriers/portenta-mid-carrier/tutorials/user-manual/content.md?plain=1#L2219-L2226

When it comes to using the Portenta C33 or H7 with the Portenta Mid Carrier, the OoB experience is to use the [arduino-libraries/Arduino_Cellular](https://github.com/arduino-libraries/Arduino_Cellular/) library. This library has been specifically developed by the Application Engineering Team @arduino/team_application_eng , to easily interact with the Pro 4G EMEA (TPX00201) and GNSS (TPX00200) modules via Arduino sketches. At the time the User Manual was developed, it may have been the case that the Arduino_Cellular library was not available.

Content and example code should be revised from [L2219](https://github.com/arduino/docs-content/blob/42e01e0f82947b8edd85b3450bea4e1de996a6ca/content/hardware/04.pro/carriers/portenta-mid-carrier/tutorials/user-manual/content.md?plain=1#L2219) to [L2343 ](https://github.com/arduino/docs-content/blob/42e01e0f82947b8edd85b3450bea4e1de996a6ca/content/hardware/04.pro/carriers/portenta-mid-carrier/tutorials/user-manual/content.md?plain=1#L2343).

sebromero, 2024-05-22:
@jcarolinares Could you please take a look? 🙏

jcarolinares, 2024-05-22:
> @jcarolinares Could you please take a look? 🙏 

To speed up things, @TaddyHC can you check this today? :)

TaddyHC, 2024-05-22:
@jcarolinares yes, we have the [PR ](https://github.com/arduino/docs-content-private/pull/672) to update the content as well just in time. Hopefully, it's ready before today!

jcarolinares, 2024-06-06:
Being this part of an internal process already I consider this issue as Done and Closed, please check related task and contact me if need it

closed 2024-06-06 by jcarolinares