arduino/docs-content #2089: Qwiic connector on the Arduino UNO R4 WiFi

baqwas, 2024-07-28:
The [documentation](https://docs.arduino.cc/tutorials/uno-r4-wifi/qwiic/) states that the following statement must be included to support the use of the second I2C interface (for Qwiic):
```
Wire1.begin();
libraryName.begin( Wire1 );
```
Unfortunately, there is no reference on how to include the access to **libraryName** in the sketch. Currently, compilation fails.

Thanks.

pedromsousalima, 2024-07-29:
Hi @baqwas,

Thank you for your feedback!
To clarify, when the documentation mentions:

```
Wire1.begin();
libraryName.begin(Wire1);
```
`libraryName` is a placeholder and should be replaced with the actual library instance you are using for your specific module or sensor like in the complete example.

closed 2024-07-29 by pedromsousalima

baqwas, 2024-07-29:
Idiot should be my middle name!