arduino/docs-content #2094: Portenta X8 RPC tutorial is not working at all

michelgokan, 2024-07-30:
I followed this tutorial to establish a "Data Exchange Between Python on Linux & Arduino Sketch": https://docs.arduino.cc/tutorials/portenta-x8/python-arduino-data-exchange/

I've tried going through the tutorial with both the latest firmware image (861 (Release arduino-91.2)) and also 746 version, on my Portenta X8, tried both with and without the mid-carrier board.

I have `py-serialrpc` running as follows:

```
bash-5.1$ sudo docker compose up
[+] Building 0.0s (0/0)
[+] Running 2/2
 ✔ Network py-serialrpc_default           Created0.2s
 ✔ Container py-serialrpc-py-serialrpc-1  Created0.1s
Attaching to py-serialrpc-py-serialrpc-1
py-serialrpc-py-serialrpc-1  | server: m4-proxy, port: 5000
```

this is the output of my `sudo journalctl -u m4-proxy -f`:

```
bash-5.1$ sudo journalctl -u m4-proxy -f
Password:
Sorry, try again.
Password:
Jul 30 15:30:23 portenta-x8-200fc209dab6fad9 systemd[1]: Started M4 builtin led forwarder.
Jul 30 15:30:29 portenta-x8-200fc209dab6fad9 m4_proxy[750]: Registering service on port  5005  with functions  map[count:5005 len:5005 tty:5005]
Jul 30 15:31:05 portenta-x8-200fc209dab6fad9 m4_proxy[750]: Registering service on port  5005  with functions  map[count:5005 len:5005 tty:5005]
```

Here is the Arduino Sketch code according to the `python-sensor-rpc` mentioned in the tutorial without any sensors:

```
#include <RPC.h>
#include <SerialRPC.h>
#include <Wire.h>

void setup()
{
    Serial.begin(115200);
    Serial.println("BME680 test on M4");
    Wire.begin();

    Serial.println("Trying to find sensor...");
    
    //RPC.bind("status", []{ return bme.sensorID() == 0x60; });
    RPC.bind("temperature", []{ return 100; });
    RPC.bind("humidity", []{ return 200; });
    RPC.bind("pressure", []{ return 300; });
    RPC.bind("gas", []{ return 400; });
    RPC.bind("altitude", []{ return 500; });

    Serial.println("Starting");
}

void loop()
{
  Serial.print("Temperature = ");
  Serial.print(100);
  Serial.println(" *C");

  Serial.print("Pressure = ");
  Serial.print(200);
  Serial.println(" hPa");

  Serial.print("Humidity = ");
  Serial.print(300);
  Serial.println(" %");

  Serial.print("Gas = ");
  Serial.print(400);
  Serial.println(" KOhms");
[... 178 lines cut ...]
Jul 30 21:53:04 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: wait for response
Jul 30 21:53:05 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: done
Jul 30 21:53:20 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: Serving 172.23.0.2:37128
Jul 30 21:53:20 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: [0 0 [116 101 109 112 101 114 97 116 117 114 101] []]
Jul 30 21:53:20 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: {[148 0 0 171 116 101 109 112 101 114 97 116 117 114 101 144] 0 0}
Jul 30 21:53:20 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: wait for response
Jul 30 21:53:21 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: done
Jul 30 21:53:36 portenta-x8-200fc209dab6fad9 m4_proxy[20457]: Serving 172.23.0.2:37130
...
```

I don't see any output in `py-serialrpc` output (I had to pip install backports.ssl_match_hostname and backports.weakref to make dockerfile compile though):

```
sudo docker compose up
[+] Running 2/2
 ⠿ Network py-serialrpc_default           Created 0.1s
 ⠿ Container py-serialrpc-py-serialrpc-1  Created 0.1ss
Attaching to py-serialrpc-py-serialrpc-1
py-serialrpc-py-serialrpc-1  | server: m4-proxy, port: 5000
```

Any ideas what's wrong? I can't even see any of the Serial outputs, this makes the whole board useless..

It's very likely that the m4-proxy is not capturing the RPC.bind from the Arduino Sketch side, because I tried a simple client/server on Python side and it was working okay! I'm not sure what's the exact problem :(

Timespansoul, 2024-11-14:
I am having the same issue on version 881. I will keep tinkering however the lack of support is ... depressing.

jcarolinares, 2024-12-02:
Hello @Timespansoul and @michelgokan 

We just released a new version of the tutorial last week that hopefully will improve this user experience:
[https://docs.arduino.cc/tutorials/portenta-x8/python-arduino-data-exchange/](
https://docs.arduino.cc/tutorials/portenta-x8/python-arduino-data-exchange/)

Please take into account that this github repository is not the **[official arduino support](https://www.arduino.cc/en/contact-us/)** and therefore resolutions can take more time than in our professional support.

Please contact my colleagues from support at: [https://www.arduino.cc/en/contact-us/](https://www.arduino.cc/en/contact-us/) who will check and try to assist you with all you need.

We will also double check the information provided here and make changes if needed, thank you for pointing out these problems!

Regards

jcarolinares, 2025-01-22:
Hello, just an update, Arduino is working on addressing this issue during the following weeks, thanks!

jcarolinares, 2025-03-13:
Hello @michelgokan I can confirm this is part of the development future plan, therefore content will be updated accordingly when improvements regarding RPC are addressed. Please contact our support team in case you need additional help!

https://www.arduino.cc/en/contact-us/

thanks!

closed 2025-03-13 by jcarolinares

michelgokan, 2025-03-13:
Hi @jcarolinares,

Thanks. Well, I was in contact with the support for a long time but they weren't very helpful in finding and resolving this issue. 

I know many people have this exact problem; you can check this thread here: https://forum.arduino.cc/t/portenta-x8-rpc-tutorial-is-not-working-on-the-latest-firmware/1286747/2

I had to switch our entire tech stack away from X8, just because of this problem. Please let me know if there is a remedy and I may be able to switch back.