arduino/docs-content #2131: Arduino Due SPI pinout in datasheet

gompper, 2024-08-26:
On page 10 and 18 of the [datasheet ](https://docs.arduino.cc/resources/datasheets/A000062-datasheet.pdf) (Modified: 14/06/2024) for Arduino Due the pins D13, D12 and D11 are labeled as SCK, CIPO and COPI for the SPI interface, which is wrong. SPI uses the dedicated pins as pointed out on page 19.

jcarolinares, 2025-03-13:
@per1234 please use this internal process to request these changes as already mentioned, if not they do not have any priority in our sprint plannings, thank you!

to check and plan actions

per1234, 2025-03-13:
@jcarolinares I did not make this request. A community member made this request.

jcarolinares, 2025-03-13:
@per1234 and you moved to this repository waiting for actions.

Being part of the Arduino team I kindly ask you to help us to deal better with these user petitions following our official request internal process so it ends up earlier in our backlog and we can help better. Thank you! 😃

jcarolinares, 2025-03-14:
@per1234 Companies have systems and workflows, and every department has a set of rules for working together and interacting with other departments. You know for sure what is our official internal request workflow to make sure the work of your colleagues and the interactions between departments work as they should be.

Those set of rules are a basic to all of us as workers, no matter our role or responsibility, so I kindly ask you to take it into consideration next time so we can work better together. That does not mean we won't check these issues, quite the contrary! I'm here to improve it indeed, but we will act better and faster following the official workflow, so again, I really appreciate if you could follow the official internal flow to do your work and the work of your colleagues better 🙏 .

Thank you!

jcarolinares, 2025-03-14:
Task is added to PX backlog, we will check the pinout and make changes if need it and update the user when done. Thanks!