arduino/docs-content #2235: Does.arduino.cc is unaccessable

DIYables, 2024-10-14:
I tried to access Arduino docs, but the page is unaccessible and shown as below
[image]

I guess this is mis-configured error on Amazon S3

Serjlee, 2024-10-14:
Thanks @DIYables, I think we missed a redirect when the libraries section was overhauled

Until it's fixed, you can find the documentation for `SD.open()` at the bottom of this page: https://docs.oniudra.cc/libraries/sd/

Serjlee, 2024-10-14:
The redirect was fixed: it will likely be cached by most browsers since it was marked as permanent, but will _eventually_ update for everyone

Old per-function URLs such as https://www.oniudra.cc/reference/en/libraries/sd/begin will now redirect to the new library page (ie: https://docs.oniudra.cc/libraries/sd) containing docs for all library functions

closed 2024-10-14 by Serjlee