arduino/docs-content #2341: Wrong rules value in installing the Arduino IDE 2 on linux instructions

Zen-cronic, 2024-12-19:
### Problem

When following the current [installation steps](https://docs.arduino.cc/software/ide-v2/tutorials/getting-started/ide-v2-downloading-and-installing/), the board can't be communicated via the port: `Permission denied on /dev/ttyACM0`.

This is because the `idVendor` attribute in the [rules section](https://github.com/arduino/docs-content/blob/2da571564a95fb270c0350793eaea098b2cd2f6a/content/software/ide-v2/tutorials/getting-started/01.ide-v2-downloading-and-installing/ide-v2-downloading-and-installing.md?plain=1#L81) is different (**2341**) from the board's info (**2a03**).

Running `lsusb` gives:
`Bus 003 Device 009: ID 2a03:0043 ... Arduino Uno Rev3`

### Solution
Users should be instructed to check the `idVendor` value by running the above command and adding that to `99-arduino.rules`. 
OR should add **2a03** instead. 

E.g., `SUBSYSTEMS=="usb", ATTRS{idVendor}=="2a03", GROUP="plugdev", MODE="0666"`

jcarolinares, 2025-03-13:
to check

oqibidipo, 2025-04-14:
It's not wrong. 2A03 is arduino.org's VID and both VIDs should have a rule.